Guizz27/DWIN_LK4Pro #1: Orizzontal version

dabox78, 2021-05-20:
Hello, is it possible to have a version with a horizontal screen, that is, like the original desuuuu with the addition of advanced functions?

Thank you so much for the wonderful work! |

Guizz27, 2021-05-21:
Which printer do you have?

dabox78, 2021-05-21:
Lk4 pro.

Guizz27, 2021-05-21:
It is a lot of work to make a horizontal version. I don't think many people turned their LK4/LK5 Pro screens to make it horizontal.
Normally I would not do this but since it could enable people with an LK1 Pro to use the same firmware.

I also have plans to have different color schemes. I like desuuuu's color scheme. 

I will work on it, but don't expect anything soon.

dabox78, 2021-05-22:
The desuuu color scheme it's a great idea!
Thanks in advance for your work!