Guizz27/DWIN_LK4Pro #2: ABL selection before printing

boromyr, 2021-07-31:
Hi, for a while now I've been thinking about a new pop-up page before each print, that lets you choose whether to start automatic levelling before each print.

**Describe the solution you'd like**
Often when printing small objects, like the calibration cube, the ABL is not of much use, especially when the Object is printed in the center, it wastes time and may make the first layer worse. By removing G29 from the startup script, it can be sent directly from the screen in case of confirmation.

**Additional context**
This idea is mainly useful in the case of bilinear leveling, for those using UBL it could be used to load the default saved mesh, or choose not to load it. Alternatively, you could start G29 J3 to align the mesh saved in memory.

Guizz27, 2021-08-29:
I am very sorry for the delay, I just saw your request. I can do a popup that gives the option to the send a G29. But it would happen before the start of the script and therefore the bed would be cold. Also turning on or off the mesh would be difficult because the script probably has a G28 which will automatically turn on or off the mesh. I will see if I could insert the G29 and or M420 after the G28 that is already in the script.

Please note that it will take a couple months before I can work on this project. So I will update  this later.

thghca, 2021-09-02:
What if there more then one homing in file?  I did it twice sometimes.

Marlin has [custom macros feature](https://marlinfw.org/docs/gcode/M810-M819.html).
Better just add new gcode that pause print and trigger macro select screen, then wait user choice. G29 and load mesh may be hardcoded macros.

boromyr, 2021-09-09:
Yes it wouldn't make sense if inserted before printing, there would be 2 homing. The custom macro might be an easier solution.

However more importantly, Desuu's firmware has been officially included in Marlin 2, and Desuu himself recently updated the code.

Guizz27, 2021-09-09:
Good idea for the macro to trigger the screen. 

It's good news that Desuuu's firmware is now in Marlin. I had to integrate new Marlin versions and sometimes it broke the code, now the code will be maintained with Marlin.