Guizz27/DWIN_LK4Pro #3: Filament change option

Baddflash, 2021-08-29:
**Is your feature request related to a problem? Please describe.**
This firmware removes the ability to swap filament during a print, the M600 command works when sent via Gcode menu so the option is enabled in marlin but not screen firmware.

**Describe the solution you'd like**
Add the ability to swap filament during print, ie issue the M600 command via button on print status screen, I have created a modified print status screen and popups for the purpose already. 

**Describe alternatives you've considered**
I have tried to resolve the issue myself but do not have the necessary programming knowledge to understand the limited documentation available regarding the DWIN screens. While this can be preprogrammed into the Gcode file it has been far more convenient to swap at my discretion. 

**Additional context**
Add any other context or screenshots about the feature request here.

Guizz27, 2021-08-29:
Thanks for the request. This feature will be in the next release which could take a few months because I'm working on another project. I want to add the possibility to change filament when printing but also from the filament main menu so we could easily change filament before starting a print. Another feature I want to add is the "prime more" on M600 before resuming the print, we could extrude a bit more filament.

As for the DWIN documentation, yes it's not obvious and not really fun to work with. I'm trying to build a screen generator. Using text files to describe the screen contents, the generator would produce the DWIN_SET folder. That will make it faster to add new features in the screen.

Baddflash, 2021-08-29:
I appreciate you taking the time to not only share your work but respond to me.

Amazing work thus far, not just the implementation of the added features but the appearance revamp, not sure why but I despise the stock longer screen firmware.

At this point the lack of filament change option is the only issue I really have, all though the z-offset feature does not seem to function either, I can live without it.

 I was going to also request the load/unload options for the filament screen but clearly you are on that.

Another option I think could be useful would be the ability to send a script to refresh the sd card, ie.. M22 M21 M20 , from the file menu to make it easier to utilize an sd card extension. I already hard wired mine to use the sense pin in the extension but this requires altering the extension and the mainboard sd card socket not exactly the most elegant solution but effective.

I have 2 of these printers so I am happy to leave one set up for testing purposes,  I have the screen mounted to allow easy access to the sd slot, clearly a necessity. It is essentially stock except for the upgraded E axis stepper driver, a 2208 v2.

Usually there is enough documentation available for me to dumb my way through this type of stuff, but after spending a week on it I am still completely lost.

I have played around with the layout a bit, adding/removing pages, buttons on those pages etc... is that something that can be done properly without altering the firmware ? or is it a hack that is likely to create issues?

I would be happy to share my work with you  and provide any assistance I can, basically anything other than the coding, lol.

Thanks again keep up the good work.
Cory

Guizz27, 2021-08-31:
Good idea for the SD card reset!

I'm a little surprised the z-offset does not work. Many options require 
to go to the eeprom menu and save it otherwise it is lost at next 
restart of the printer. I'll make sure to test it before the next release.

For DWIN, you can remove buttons without having to modify the Marlin 
firmware. Some pages but not all could be removed without modifying 
Marlin. Similar for moving buttons to different page, it will work for 
some but not all. This is because part of the screen navigation is done 
in Marlin.

You can't move displayed values to a different screen. For example, you 
cannot show the extruder temperature on a page that does not currently 
have the extruder temperature. In the Marlin code there is a mapping on 
which variable to send when a page becomes active. You would need to 
modify the file 
Marlin/src/lcd/extui/dgus_reloaded/definition/DGUS_ScreenAddrList.cpp in 
Marlin if you want to display a value on a different screen.

There is no way to send gcode directly from the screen right now. Like 
if you want to add a button that sends a M600 for example. When you type 
a gcode command in the screen, it saves the command on the printer, then 
when you "Execute" it, it sends a command to execute the stored gcode. 
You cannot hack this process without modifying Marlin because the 
"execute gcode" command verifies the printer is not printing.But if you 
pause first, then you could hack it by having a button to send the M600 
command and another button to execute the gcode, or pause, then go to 
the gcode screen page and manually enter the M600 command. Please note, 
there might be other checks somewhere that could prevent these hack.

Normally, the screen would send a code (number) to Marlin and inside 
Marlin that code is interpreted and then it inserts a gcode or calls a 
function. For example, homing the printer sends a code to Marlin and 
Marlin inserts a G28 in the command queue. But if you change the 
temperature, it calls a function inside Marlin to set the value and will 
not use M104.

Thank you for using and testing the firmware and also for providing 
feedback.

Baddflash, 2021-08-31:
Thanks again for the info, it helps clarify a number of things.

I should clarify the Z-offset may work however the numpad screen doesn’t load properly, ok and cancel buttons are not visible and the screen stays locked there, I tried pressing where the buttons should be with no reaction.
I have only tried it a couple of times but had the same reaction each time. I have reset the processor each time so I am not sure if it would return to the print status screen if left alone. The keypad seems to function fine with exception of the ok and cancel buttons.

I can test it’s function further to give a more accurate assessment.

One of my hacks seems to work than as I did add the bed and extruder temps to the print finished screen and they work well. I have found issues on other pages of the temp only updating when the page is opened, not continuously.  I will try updating the screen address file and see if that corrects the issue.

A few small things I think could be tweaked, the z-axis doesn’t raise after leaving the manual leveling screen, or when a print is cancelled.
The other is the move screen requires homing first which is unnecessary and can defeat it’s use at times.
Small issues but after 800+ hours with the stock firmware they are missed.

Happy to help with testing prior to your next release.