Guizz27/DWIN_LK4Pro #4: it doesn't work

Teius92, 2021-08-31:
**Describe the bug**
I put all files in the sd ad waited that the installation was done, but once i restarted the printer i only could see "resetting eeprom" (or something similar) in the bottom right angle. The screen is unusable and if i touch here and there appears some number with "mm".

**To Reproduce**
Steps to reproduce the behavior:
Copy all the files in a formatted sd and put it in the sd reader of the screen.
Turn on the touchscreen and wait till yhe end.
Restart the screen.

**Expected behavior**
that it work.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

Guizz27, 2021-09-01:
I'm sorry you are having issues. Did you also update the printer firmware from [here](https://github.com/Guizz27/Marlin---Longer-LKx-Pro/releases)?

Do you see the boot screen with the word LONGER and the Marlin icon when you turn on the printer?

Do you have the LK4 Pro? And how long have you had it?

thghca, 2021-09-02:
> Copy all the files in a formatted sd and put it in the sd reader of the screen.
From releases? Sources has wrong images orientation.

Also kernel update requred for some screens.

Teius92, 2021-09-02:
Hi, i have the printer (Longer LK4 PRO) since the 1st of August. My printer
is stock, no custom parts. The screen is the stock one too.
I just copied all the files in the SD and put it in the SD screen slot. The
installation went fine (it said the end message) but when I start the
printer i only can see clearly the Marlin logo. After that i only see a
violet distorted image with a green phrase abouth the EEPROM. If i touch
something seems to happen, but i don't know what. The last time appeared
some numbers. I didn't modified anything, just compiled marlin and uploaded
it. The same for the screen. Thanks a lot for the help.

thghca, 2021-09-02:
> fter that i only see a violet distorted image

Flash screen firmware files from releases https://github.com/Guizz27/DWIN_LK4Pro/releases
Not from sources.

Teius92, 2021-09-02:
Ok, i'll try and will tell you if it work. Thanks, i don't know a lot of
GitHub, so i downloaded the wrong file.

Teius92, 2021-09-02:
Excuse me, one last thing. I have to download even the release of the
firmware of the 19th may, i suppose.
It say to use M503, but i have stock parameters.do you think I need to use
it or not?

thghca, 2021-09-02:
M103 just print current settings in terminal for you copy it somewhere.
Bare minimum for you with stock printer is comment #define LK4_Pro_BLTOUCH.

But I recommend go trough whole config (it's big, yes) and check every parameter and ajust if nessesary. Then compile and flash.

Teius92, 2021-09-02:
Ok, i will read and check every parameter if it's necessary to define or
not. Thanks a lot, I'll inform you if it works (I've not much fee time, so
it could take a little long). Have a good day.

Baddflash, 2021-09-04:
I noticed when trying to return to stock screen firmware I had to rename the 13.bin and 14.bin files to english, ie..  13_touch.bin and 14_variable.bin or the stock firmware will not overwrite the config files from this screen firmware and you would get a nonfunctional mashup of the two.

Teius92, 2021-09-04:
Ok, so i need to rename those files to go from your custom screen firmware
to the stock one. Thank you!

Baddflash, 2021-09-04:
Not mine just another user. yes that will get the touchscreen working again as long as the printer firmware is stock as well. Both have to be matched to work properly.

Guizz27, 2021-09-04:
I just want to check again what you copied on the SD. did you take it from the release page in the assets section? You should take the DWIN_SET.zip from the assets in https://github.com/Guizz27/DWIN_LK4Pro/releases

If you take the DWIN_SET from the main github (code) page it will not work because the images are not rotated. If you want to use the source code, you have to execute the BuildRelease,bat file which will create a folder called "Release" with a subfolder called DWIN_SET, it would be that one you have to copy to the SD card. But instead, it is easier to copy it from the release page. 

Here is a direct link to the zip file https://github.com/Guizz27/DWIN_LK4Pro/releases/download/1.0-RC3/DWIN_SET.zip. Try with that zip file.

I had distorted images before and it was because the images were not rotated in DWIN_SET.

Baddflash, 2021-09-04:
It sounds like he never installed the proper printer firmware to go with the screen firmware. The stock longer screen firmware will not properly reinstall after this firmware unless those 2 files are renamed in the longer DWIN_SET.

Guizz27, 2021-09-04:
This is interesting because according to the DWIN documentation, the file naming is 13*.bin or 14*.bin.  Maybe because I named the files 13_touch.bin that is now has to be named like that. Thanks for this information.

Serjkustarev, 2021-09-04:
Hello, i try it and have same, don't work. 

I use stok longer lk4pro, comment bltouch, upload FW by VSCode, check it by pronterface, he told marlin 2.0.9. Use M502, M500 to start 

Drag and drop realese version FW for screen on microSD and put it screen, turn on, screen show blue, and tell he copied some files, but still 0.3.1 -DW version

Baddflash, 2021-09-04:
the stock screen firmware will only work with marlin 1.1.9 and it has to be their version. you can not make it work with marlin 2.0.x or another version of marlin 1.19 , the printer firmware is customized to work with the custom screen firmware.

Baddflash, 2021-09-04:
The same is true for this screen firmware, it has to be used with the printer firmware he customized, you can not simply load whatever version of printer firmware and expect the two to work together. This is true for every printer that uses this style of touchscreen.

Serjkustarev, 2021-09-04:
> the stock screen firmware will only work with marlin 1.1.9 and it has to be their version. you can not make it work with marlin 2.0.x or another version of marlin 1.19 , the printer firmware is customized to work with the custom screen firmware.

well i trying to instal this custom screen frimware and can't, maybe i  need to use some else frimware first? like unlock bootloader on phones

Baddflash, 2021-09-04:
If you want to use this screen firmware, download the version of marlin 2.0.x he links to . compile marlin without any changes and upload with cura 15.04. upload the his screen firmware as it is and how you have been doing. using both of these it does work fine.

Serjkustarev, 2021-09-04:
Now i try without modify FM (was commented Bltouch) result is same, don't update frimware on screen. 

Try 2 times, without M502, M500, and with it.

Baddflash, 2021-09-04:
I have been playing with this firmware and the dwin screen for a couple weeks now. The files he uploaded will work fine without modifying anything. I suggest you start with fresh copies because other than the issue I described when returning the entire printer back to stock firmware it should be simple. You could maybe try the arduino ide to compile the printer firmware instead.

Guizz27, 2021-09-05:
@Serjkustarev 

The printer firmware seems to be fine from your screenshot.

But no files were copied in your screen firmware (all 000). 
On the SD you should have a folder called DWIN_SET. Inside that folder you should have many files, like 000_boot.bmp, 0_DWIN_ASC.HZK, ...

If you do and it does not copy files you may try to update the kernel. Please note I never did it. I'm not sure how to revert it if it breaks something. To update the kernel, you have to download the 2 .bin files here https://github.com/Desuuuu/DGUS-reloaded/tree/master/extras/core_firmware and put them on the SD in the DWIN_SET folder.

Baddflash, 2021-09-05:
I agree there should be no need to mess with the screens kernel. Sorry I hadn’t noticed the screen shot, and that no files transferred.  Are you placing the dwin_set folder on the root of the sd card? You should see all the images of the firmware flash on the screen as it uploads.

Serjkustarev, 2021-09-05:
> I agree there should be no need to mess with the screens kernel. Sorry I hadn’t noticed the screen shot, and that no files transferred. Are you placing the dwin_set folder on the root of the sd card? You should see all the images of the firmware flash on the screen as it uploads.

yes, i try fresh sdcard, formated and totaly clear, only dwinn_set folder (62 files), result same. Tryind old version, same. i add pic of my screen, maybe it's different, printer is new (1.09) 

i can try to update kernel, i just need put it on sdcard?

[image]

Baddflash, 2021-09-05:
That looks to be the same model screen as my printers, so this should work. It seems like the screen is not seeing the folder. What size sd card are you using, these screens will not read above a 32gb card, it must be formatted as fat32 as well.

Serjkustarev, 2021-09-05:
> That looks to be the same model screen as my printers, so this should work. It seems like the screen is not seeing the folder. What size sd card are you using, these screens will not read above a 32gb card, it must be formatted as fat32 as well.

16Gb and 8GB, FAT32

Baddflash, 2021-09-05:
Unless the folder is named incorrectly I cannot think of why this doesn’t work.

Baddflash, 2021-09-05:
It is possible to disable the sd card slot however I have no idea if it would ignore the sd card altogether or attempt an upload as yours is doing. I also can not think of any reason Longer would do this and prevent any future updates.

Serjkustarev, 2021-09-05:
find, it was on sd card. need be formatet fat32 4096 only! 

Trying it, and it's working good

Baddflash, 2021-09-05:
It’s amazing how much time could have been saved by simply reading the instructions. This was all clearly spelled out by Longer in the instructions included with their firmware on how to update the firmware.

Serjkustarev, 2021-09-05:
> It’s amazing how much time could have been saved by simply reading the instructions

think so, but it can be added in instuction, all read instructions and don't read ton of text

Baddflash, 2021-09-05:
This firmware is someone’s personal project they spent a lot of time on and were good enough to freely share it with others, with the only expectation that you know how to use it.

It seriously never occurred to you that maybe you should learn how to update the firmware before actually trying too , failing, than complaining that it doesn’t actually work.

Instead you want us to play guessing games just to find out you did not read the instructions and therefore did not format the sd card correctly .

Consequently this entire thread was a waste of everyone’s time and should never have existed, good work.

Guizz27, 2021-09-05:
I'll take the blame as I haven't provided instructions on how to update the firmware. I added some basic instructions in the readme but I will add better instructions with images later when I have more time to put on it.

Thanks Baddflash for helping with this issue.

Serjkustarev, 2021-09-05:
> This firmware is someone’s personal project they spent a lot of time on and were good enough to freely share it with others, with the only expectation that you know how to use it.

First of all, I am grateful to the author for his work.

But you write in such a way that it burns with the fact that someone inexperienced uses this software, and does not know the rules for its use.

Perhaps you work in this area and know some of the subtleties. But this does not mean that you are smarter than everyone, and it does not give you the right to taunt inexperienced users.

Baddflash, 2021-09-05:
I genuinely tried to help you through this issue, offering any help I could. 
However I did not assume that at no point did you ever attempt to look for instructions and I would have never assumed you would actually ignore them. 

[image]

You were looking at this folder, while we were trying to help you as you were struggling with installing the firmware????? Hmmmm

I apologize if pointing out the obvious facts  hurt your feelings, I forget a lot of people have issues with that these days.

Guizz27, 2021-09-05:
I am closing this issue. Comments are becoming too aggressive for my taste. 
thanks again @Baddflash for providing support.

@Teius92, I'm pretty sure the problem you had was the wrong DWIN_SET folder was used. If you get a chance to try again and it still doesn't work please open a new issue here on github.

Like I said, I should have given my own instructions on how to install the firmware. Or at least reference the official Longer instructions. On my side I learned a few things. This is really the first time I provide a piece of software, usually I just do stuff for myself.

closed 2021-09-05 by Guizz27

Teius92, 2021-09-06:
Thanks a lot to all for the help. The firmware works well and all the problems i had, were caused by my mistake (source file download). Sorry for the inconveniences. Have a good day.

Teius92, 2021-09-06:
I didn't know how GitHub work since you told me that there is the section
"releases" so, i just downloaded all the packages with the green button on
the right that is in the upper zone (just a little under the main title).
I'm sure that i didn't download the files that you recommended me to
download from the "releases" section. I will try (maybe tonight, I'm in
Italy and here are 17:17) to flash both firmwares (of the right releases)
and see what will happen. The only thing that scares me (if we want to say
so) is that i have to rename some files to make all work (I'm not very able
as a programmer, i think you understood that). Just one last thing: have i
to rename the files to install the custom FW and have i to rename those
back to install the stock one? Sorry, but I'm a bit dumb in such things.

Guizz27, 2021-09-06:
> Just one last thing: have i to rename the files to install the custom FW and have i to rename those back to install the stock one?

You should not have to rename the files to install the custom FW. 
Apparently you need to rename the files 13.bin -> 13_touch.bin and 14.bin -> 14_variable.bin in the stock FW before you install it over the custom FW. I never had to do it so I cannot confirm but it should be easy to rename these 2 files if you have to.