Guizz27/DWIN_LK4Pro #5: About localisation and unicode support

thghca, 2021-09-02:
Marlin has no proper support of unicode filenames but maybe useful for localisations.

Dgus text controls has encoding=unicode. It accepts utf16.
caveat: first char byte can be 00, so no null-terminate C-style strings.

Guizz27, 2021-09-03:
I haven't looked at the Marlin support for utf16 yet. I wasn`t planning to add support for non-latin languages. I will take a look at what is supported in Marlin when I get to work on it in a few months.

thghca, 2021-09-08:
No luck with unicode. =(
But i successfully made screen display Russian with 8bit ascii mode using win1251 encoding.

Guizz27, 2021-09-09:
Interesting. To work with other languages as well, the screen firmware will have to be setup correctly and the transfer of the text to the screen as well. I know this part only supports 8bits for now.