Guizz27/DWIN_LK4Pro #7: Z-offset print tune screen

Baddflash, 2021-09-07:
**Describe the bug**
When trying to adjust Z-offset during printing the keyboard is not completely visible, can not see use cancel or ok buttons, can not return to tune or print status screen.

**To Reproduce**
During the start of a print access the tune menu and try to adjust z-offset.

**Expected behavior**
The entire screen should change and show the full keyboard. 

**Additional context**

The issue is the location information for the z-offset variable data input. 

These changes corrected the issue for me.

The main show location = 200-133

Also under the keyboard settings 

Keyboard area = 0-0  272-480

and show location = 0-0

Baddflash, 2021-09-08:
Not sure where to post this, no IM's here. I created a firmware tutorial, I uploaded the pdf to my github.

https://github.com/Baddflash/LK4-Pro-Firmware-Tutorial

Feel free to link to it or integrate it with your project.

Cheers

Baddflash, 2021-09-08:
Also curious if you would be ok with me uploading my edited version of your screen firmware?

Guizz27, 2021-09-09:
> Not sure where to post this, no IM's here. I created a firmware tutorial, I uploaded the pdf to my github.
> 
> https://github.com/Baddflash/LK4-Pro-Firmware-Tutorial

Thanks a lot! I will link it. Note that I use VSCode with platformIO to build and upload. It's just that I may not be able to help people trying to build with Arduino IDE if they have issues.

Try to create a pull request for your changes.

Baddflash, 2021-09-09:
No worries. I can help out there if you need me too. I stuck with what I knew and it’s worked for me. 

I’m still pretty new to the scene as I only got my first printer in July, so I’m still learning the specifics.

But given state of things I’ve had the time to condense what would have probably been years of use into a couple months, lol. Between the 2 I have (and the 3rd that was returned, go Longer QC) I’m getting close to 1k hours. 

My first one
[image]

jgbrown54, 2022-04-17:
Has this fix been included in the DWIN_SET yet? Let me clarify my issue. If you go into the Z-Offset screen from the Print Status screen, Using the up and down arrows works fine for the Z-Offset. But if you open the keyboard to enter a specific value, there is no way to return to the Print Status screen.

Guizz27, 2022-04-22:
The issue is fixed but I did not create a release. If you get the repository's code and execute BuildRelease.bat it will create a folder called "release" and inside there will be the DWIN_SET folder that has to be copied to the SD card. This is not the same as the DWIN_SET that is in the repository.

Guizz27, 2022-06-24:
The fix is now included in the DWIN_SET in Release 1.1. 
https://github.com/Guizz27/DWIN_LK4Pro/releases/tag/1.1

closed 2022-06-24 by Guizz27