Guizz27/DWIN_LK4Pro #8: Gcode(screen) not working

darky552435, 2021-12-11:
after trying to use gcode on screen its said update to dwin gui\os. i dont see anything then google it, can u say what i should do

darky552435, 2021-12-11:
also forgot to mention that auto home isnt work then i press start. bltouch is working

Guizz27, 2021-12-11:
where did it say to update the dwin gui\os ? Is it possible to attach a picture of the message?
Which gcode command did you try to run?

darky552435, 2021-12-12:
btw i cant even run 1 gcode command or even write it. 
happen then i click gcode icone
https://pin.it/2HI0xsM

Guizz27, 2021-12-12:
Thank you for the picture and the explanations. It looks like your OS in the screen firmware is an old version.

You can check your gui and os version but going into Settings then About. Then tap 10 times anywhere on the screen, this should display a screen with the version numbers. GUI should be at least version 48 and OS should be at least version 33. See attached picture as a reference.

Download the 2 .BIN files from here https://github.com/Guizz27/DWIN_LK4Pro/tree/main/extras/core_firmware
Add them to the DWIN_SET folder on your SD card, then follow the same procedure as before to upgrade the screen firmware. This will install a newer OS on the screen. You only need to update these once, so remove them from the SD card when done.

Please note that this is not a process I had to do before. I'm just giving you the instructions I found on Desuuu github https://github.com/Desuuuu/DGUS-reloaded/wiki/Flashing-the-T5-GUI-and-OS-firmware. 

Let me know how it goes and if it fixes the issue. 

[image]

darky552435, 2021-12-12:
well, now its working, thank u so much (for testing run g28,29)

closed 2022-01-02 by Guizz27