Guizz27/DWIN_LK4Pro #9: Question

jgbrown54, 2022-04-14:
This is more of a question than an issue.

A friend tried to update but he couldn't get the DWIN_SET to load so he fell back. He said he had to use .TFT files. He is currently on 0.3.2-DWTJ. Does that make sense?

jgbrown54, 2022-04-15:
When we try to update on a display showing 0.3.1-DW, using DWIN_SET, it doesn't update. The old screens remain.
[image]

jgbrown54, 2022-04-15:
I finally figured this out. Both of these systems were running Windows 11 Home Edition. It does not allow formatting an SD card with an Allocation unit of 4096. I had them go into a command prompt and enter -- >Format D: /FS:FAT32 /Q /A:4096 where D: is the drive where the SD card is mounted.

Guizz27, 2022-04-16:
Good catch about Windows 11. 
Did it work for your friend with a DWTJ screen firmware after formatting the SD card correctly? I'm curious because that format uses a tft file but maybe it also support a DWIN_SET folder.

jgbrown54, 2022-04-16:
I do not know. The guy that used the .tft file, went back to the old FW and he was ticked that he went through all the trouble to load the new FW and it didn't work.

jgbrown54, 2022-04-21:
Sorry to reopen this in the issues section. I'm new to Github and I don't see anywhere to ask questions.

The guy that had the DWTJ tried again and made sure that he had the SD formatted properly. It still wouldn't load the DWIN_SET. Couold he maybe load the TFT files from here? What is the process to load the TFT files?

I have been helping people on the Longer3d Facebook page load your firmware and screens. Most just have issues following instructions.

Guizz27, 2022-04-22:
I don't know much more about how Github works for issues and comments either.

I don't have a TFT file and I don't know how to produce one. I'm afraid that only people with a DW screen (either no letter after the screen fw version like 0.3.1 or the letters DW like 0.3.1DW) will be able to use the firmware from here.

My guess is the OS on the screen has to be updated but this is just a guess. I have the OS here https://github.com/Guizz27/DWIN_LK4Pro/tree/main/extras/core_firmware that could be used to update the OS on the screen and then accept the DWIN_SET folder. Again, this is a guess. I would not recommend the guy to update the OS unless he has a copy of the original that he can put back.

BTW, thanks for helping people, this is very appreciated.

Guizz27, 2022-04-22:
People who have a DW screen and can't enter gcode commands on the screen have to update the OS by copying the 2 files from the link above to the DWIN_SET folder on the SD card. More detailed instructions here: https://github.com/Desuuuu/DGUS-reloaded/wiki/Flashing-the-T5-GUI-and-OS-firmware

jgbrown54, 2022-04-22:
The guy that had the 0.3.2-DWTJ display just blew up his motherboard. So he going to a different motherboard screen combo. He has offered me his old screen for nothing. It will give me a chance to see if I can figure out a solution. Would it be possible to get an email or messenger address for you or should I just keep opening this issue. I promise not abuse it but I might have some questions. Should I add the files mentioned above to the DWIN_SET that I am providing to everyone or just for people having issues entering gcodes?

Guizz27, 2022-04-22:
The 2 files should only be added for people with issues. When you get the DWTJ screen you can try to add those 2 files to see if it will help.
To keep asking questions you can try the "Discussions" tab at the top in GitHub. I enabled it. Not sure how it works. I think it can help to keep discussions in github for future reference and reference by other people.

I will PM you on messenger, so you can also contact me as well for very specific things.

Guizz27, 2022-07-11:
Closing this issue. We learned about the SD format parameters and we now know that T5L, DWJT and DWJTB screens cannot be upgraded.

closed 2022-07-11 by Guizz27

Gilgul46, 2023-03-07:
> Closing this issue. We learned about the SD format parameters and we now know that T5L, DWJT and DWJTB screens cannot be upgraded.

Why can't they be updated? What did you learn about the SD format parameters that made you come to the conclusion that T5L, DWJT and DWJTB screens cannot be upgraded?
Also, do you know where can I get the DW screen? If I can't update mine, I just have to replace the screen, but I can't really find clearly where to buy it 😅

Guizz27, 2023-03-07:
For the upgrade to work on the DW screen, the SD card should be max 16GB and be formatted in FAT32 with 4K cluster size.
To upgrade a non-DW screen, I need to use a different software to create the firmware as the file format is not compatible between those screens. Since I only own a DW, I cannot develop and test for other screens.

The screen you want is DMT48270C043_06WT. There is one on aliexpress https://www.aliexpress.com/item/4000957547147.html?spm=a2g0o.cart.0.0.9a6338daJB8zaz&mp=1 (has to be 06WT) but it does not come with the white connector on the back. You could probably use the one from the screen you have if you can unsolder it and solder it to the new screen.

James Brown (@jgbrown54 )  sometimes has DW screens that he sells with the white connector.

Gilgul46, 2023-03-07:
DUDE THANK YOU SO MUCH!!!!
I really appreciate everything, and if you're in Europe I can send you my current screen, so you can work on it if you wanna xd 
I already gotta buy a new one, so I don't really mind giving you my current one hahaha, especially to the guy who works on this cool updates for the Longer LK4 pro (and I use bl-touch too hahaha) ;)

Gilgul46, 2023-03-07:
> For the upgrade to work on the DW screen, the SD card should be max 16GB and be formatted in FAT32 with 4K cluster size. To upgrade a non-DW screen, I need to use a different software to create the [...]

DUDE THANK YOU SO MUCH!!!!
I really appreciate everything, and if you're in Europe I can send you my current screen, so you can work on it if you wanna xd
I already gotta buy a new one, so I don't really mind giving you my current one hahaha, especially to the guy who works on this cool updates for the Longer LK4 pro (and I use bl-touch too hahaha) ;)

Guizz27, 2023-03-08:
HaHa! I'm in Canada :).
Looking at the pictures of your screen, the white connector would not work on the DW screen because you have 4 pins and the DW has 6 pins with only 4 used but the used ones are not consecutives. So you need to replace the plug on the cable as well.

Gilgul46, 2023-03-10:
> HaHa! I'm in Canada :). Looking at the pictures of your screen, the white connector would not work on the DW screen because you have 4 pins and the DW has 6 pins with only 4 used but the used ones are not consecutives. So you need to replace the plug on the cable as well.

DUDE I GOT SO LUCKY, LOOK: https://www.ebay.es/itm/115618664377?mkcid=16&mkevt=1&mkrid=1185-127638-7840-0&ssspo=eMhAl0lETbK&sssrc=2349624&ssuid=cO_U2_dhSmi&var=&widget_ver=artemis&media=WHATS_APP 
It's that right??? PLUS THE CABLE HELL YEA!!

Guizz27, 2023-03-10:
Yes that's the right thing!!

Gilgul46, 2023-03-19:
Hey @Guizz27, just wanted to tell you that after doing some things with Proterface it told me that the old screen is DWIN-T5, and after a simple search I found this GitHub rep: https://github.com/ihrapsa/T5UIC1-DWIN-toolset. I don't know if it helps you in any way, but I hope it does :)

Guizz27, 2023-03-20:
Thanks for the link. I will take a look.
T5 is the same one I have. Is the "old screen"  the used one from ebay or the original one that came with the printer? If you have the original screen dismounted, could you take a picture of the back showing any number that could look like a model number including the chips models?

Gilgul46, 2023-03-20:
The "old" screen is the newer version, aka the 4 letter combination
version. The Ebay screen is still in shipping, but I'll unmount my current
one and send pics!

I found it curious that pronterface says the new sort of screens is a DWIN
T5

Gilgul46, 2023-04-07:
Hello! 
The screen got lost in shipping and I wanna cry, but here's the screen :)
[image]

Gilgul46, 2023-04-07:
> Thanks for the link. I will take a look. T5 is the same one I have. Is the "old screen" the used one from ebay or the original one that came with the printer? If you have the original screen [...]

Here is also a pic of a zoom on the chips:
[image]
If you want any other pics, tell me! :)

Gilgul46, 2023-04-07:
Here it is from the back: 
[image]

Gilgul46, 2023-04-07:
Here it is from the back: 
[image]

And another pic just in case:
[image]

Gilgul46, 2023-04-08:
After reverse image search in AliExpress it looks like this:
€ 14.68 | New listing TJC4827X243_011 X2 series 4.3 inch IPS HMI serial screen resistive touch LCD display human-computer interaction
https://a.aliexpress.com/_Ewa1OLN

Gilgul46, 2023-04-17:
> Thanks for the link. I will take a look. T5 is the same one I have. Is the "old screen" the used one from ebay or the original one that came with the printer? If you have the original screen [...]

Hello! any way I can help?

Gilgul46, 2023-04-17:
I still didn't receive a response regarding everything haaha

Guizz27, 2023-04-18:
Sorry, your screen is not supported at this time. You have a TJC... screen and the chip says T7. The link you reference on aliexpress may be the same as the one you have or very similar.

The good news is the cable for your screen is a 6 pin connector.  If you buy the one on aliexpress from the link I gave you and you can unsolder/solder the connector from your old screen it would work.

If you are not comfortable to solder the connector, I can order the screen, install a connector and ship it to you but it would cost at least $30 US + shipping to you which could end up cost you about $45 US total or even more depending where you are.

Gilgul46, 2023-04-26:
Thanks for the offer, but I ended up doing it myself and HOLLY MOLLY IT'S SO MUCH BETTER THAN STOCK. Like, finally, I don't need my computer whenever I want to calibrate the BL-Touch, or input any specific short G-code!!!! thanks so much, truly!!! I just hope you'll be able to make it so anyone with any screen can enjoy it one day :)