Guizz27/DWIN_LK4Pro #14: Initialization Of ABL Table

jgbrown54, 2022-05-23:
From complaints I have received from people I have shared your code with, it appears that the ABL table is not initialized on a new FW/DWIN load. After a fresh install, ABL does not launch when selected from the screen. After issuing a first G29, it then works as it should. I wasn't sure if this was a DWIN issue or a Marlin issue.

Guizz27, 2022-05-26:
One thing I know is when you upgrade from the original Longer fw to this one, the data stored in the EEPROM is no longer valid. This is a behavior of Marlin. So this is probably normal that the table is not initialized (all 0s).
But, you should still be able to start the ABL from the screen without having to manually run a G29.

I will see if I can find the issue.

Guizz27, 2022-06-27:
This should be fixed in the new release https://github.com/Guizz27/Marlin---Longer-LKx-Pro/releases/tag/2.0.9.4-GC-1.1

closed 2022-06-27 by Guizz27