Guizz27/DWIN_LK4Pro #15: Problem upgrading from previous horizontal version

FulvioSpelta, 2022-05-25:
**Describe the bug**
I've update my U30 Pro display starting from an horizontal version. The result is that the bitmap are "portrait" but all other references such as message positioning, temperature positioning in home screen and button are "landscape". See attached images (during upgrade and the home after upgrade):

[image]
[image]

**To Reproduce**
Upgrade fw from a horizontal version (I've done the upgrade more times)

**Expected behavior**
Full portrait

FulvioSpelta, 2022-05-25:
**Workaround**

I've installed the original Alfawise display fw (that is vertical looking) and that package has rotated the configuration of the display, further i've installed this fw and it's all ok.
[image]
[image]

Guizz27, 2022-05-26:
Hi @FulvioSpelta, thanks a lot for investigating and finding a workaround. I will look to include the file T5UID1.CFG in a near future.

Guizz27, 2022-07-11:
I now include the file T5UID1_reset90.CFG which will reset the orientation and will also erase previous data in memory to make sure there is nothin left that could interfere with this firmware.

closed 2022-07-11 by Guizz27