Guizz27/DWIN_LK4Pro #21: Cannot set hotend temp correctly, temp range incorrect for PID tune

babblephish, 2022-07-09:
**Describe the bug**
Cannot set hotend past 100 for PID tune

**To Reproduce**
Steps to reproduce the behavior:
1. Go to PID screen
2. Click on temp
3. try to enter anything over 100
4. goes back to default 200 setting in Marlin

**Expected behavior**
Be able to set temp to max temp

I fixed this in my fork by changing the keyboard range min/max to 5/275 on the PID Tune screen. What you might want here really depends on what you have in your config.

I used DGUS V7391 to make the change, not sure how compatible it might be with whatever vesrion you may have used.

Guizz27, 2022-07-09:
Thanks for the bug. I will fix it.
I could also have Marlin set the limit on the screen based on the confg. Same for bed temp.
I used V7381, should be compatible.

babblephish, 2022-07-09:
Yes, removing the limit should work. I had an odd issue with the DGUS tool where it would just go and set the number of digits to zero for some reason when I did that. I don't "think" it was me doing it. I would load the new screens and suddenly could not enter anything. So, I just left it with the limits.

Guizz27, 2022-07-11:
I removed the limit and tested it. It works fine. I can set a value above 200. A value above 260 will set it to 260 which is I think a max of 275 minus a 15 degrees for overshoot.

I updated my repository with the fix.

closed 2022-07-11 by Guizz27