Guizz27/DWIN_LK4Pro #22: Replace split QWERTY keyboard used for Gcode entry with logical keyboard for non-touchtype keyboard.

jgbrown54, 2022-09-15:
Attached is a zip file copied from the latest DWIN source for version 1.2. The only changes are the replacement of screens 200 and 201 used for entry of Gcodes. This configuration has been tested.
[DWIN_LK4Pro-main Modified.zip](https://github.com/Guizz27/DWIN_LK4Pro/files/9572278/DWIN_LK4Pro-main.Modified.zip)

iw5edv, 2022-11-12:
It doesn't work on my Longer LK4pro. Doesn't load all screenshot images correctly

jgbrown54, 2022-11-13:
Do all of the other screens display correctly? Only the keyboard screen for Gcode entry was changed.

iw5edv, 2022-11-13:
Sorry, my fault I guess I was loading the folder with the images not rotated 90 degrees.
As soon as I can try again.

jgbrown54, 2022-11-13:
Why are you rotating them? You shouldn't have to. Are you formatting the SD card per the instructions? Did you verify that you have the DWIN display and NOT the DWTJ (Techonics) display?

Guizz27, 2022-11-13:
Make sure you run BuildRelease.bat which will create a release folder with the proper file names and image rotation.