Guizz27/DWIN_LK4Pro #23: Piccole modifiche e migliorie

iw5edv, 2022-11-12:
Sorry for my bad English.

Hi, since you are planning to work on a new version I give you some points to work on.

In the Setting-> Screen, the back button points to the wrong page.

The line at the end of the pages where you see the printer messages, the font should be increased and possibly in bold.

Very important is the reload reset of the SD from the screen (for remote reader)

In the GCODE page the keyboard is incomplete.

I am trying to edit / correct but the DWIN displays are not straightforward and poor documentation.
I have big problems compiling because the images "break" and the screen becomes unusable

Guizz27, 2022-11-13:
Thank you for your feedback. I will add tour points to my list for the next release.
I'm curious about the GCODE page not being complete. Could you give me more informations?

Agreed, working with the DGUS software is not easy. Once you make changes, run the BuildRelease.bat file, it will create a release folder with the proper file names and image rotation.

iw5edv, 2022-11-18:
I started working and understanding how DGUS works.

The GCODE keyboard is correct... It was me who couldn't find the keys. ;)